andre1araujo/YOLO-on-PYNQ-Z2 #3: A problem during Project configuration

menger02, 2024-04-17:
Hello, I'm very sorry to disturb you. 
When I write petalinux-config -c rootfs ,it has the problem like this. Can you help me?
[image]

andre1araujo, 2024-04-17:
Hello!
You are not bothering at all. Fell free to ask any questions!

About the problem you are facing, it can be due to a lot of reasons. I suspect it is something simple. I looked at [this forum post](https://support.xilinx.com/s/question/0D52E00006hpWZjSAM/petalinux-error-failed-to-create-workspace-directory?language=en_US) where someone was having a similar error. If you scroll down, there is this response:

[image]

What they suggest is that **there was a "CTRL+C" command** somewere alog the process. And according to your screenshot, you did exactly that. You can do what they say on the comment above (kill processes) or you could just **close petalinux or even restart Ubuntu and then open the Petalinux project again and execute the command you did**. If it does not work you could start the Petalinux project again and this time make sure to not "CTRL+C" in the middle of any process.

If that does not work, the **problem might be related with the permissions of the folder/ path** your project is located in. I don't think this is the problem but you could try the next command: `chmod 777 Desktop/dpu_integration/pynqz2_dpu`
This command gives read, write and execute permissions. 

I hope this helps! If the problem pressists be sure to come back to this post! 
If you are able to solve the problem, I request that you come back and tell us so that this post can be usefull to another person that might get stuck on the same problem!

menger02, 2024-04-18:
Hello, thank you for your reply. But unfortunately, it's not as you said. Through the website you shared, I learned that the cause of the error can be understood through the build/config.log file. I found that the problem lies in the     echo 'IMAGE-INSTALL-append="dpu".     After changing it to    IMAGE-INSTALL-append="dpu", this error disappeared.
 But I encountered a new problem again, as shown in the second and third images. I guess it was caused by the colon after the DPU, but after I made the modifications, I found that it was not like this, and after running it, it changed to its previous appearance. Do you know the specific reason? Or is it that my previous modifications were not correct?
[image]
[image]
[image]

menger02, 2024-04-18:
> Hello! You are not bothering at all. Fell free to ask any questions!
>
> About the problem you are facing, it can be due to a lot of reasons. I suspect it is something simple. I looked at [this forum [...]

Hello, thank you for your reply. But unfortunately, it's not as you said. Through the website you shared, I learned that the cause of the error can be understood through the build/config.log file. I found that the problem lies in the echo 'IMAGE-INSTALL-append="dpu". After changing it to IMAGE-INSTALL-append="dpu", this error disappeared.
But I encountered a new problem again, as shown in the second and third images. I guess it was caused by the colon after the DPU, but after I made the modifications, I found that it was not like this, and after running it, it changed to its previous appearance. Do you know the specific reason? Or is it that my previous modifications were not correct?
[image]
[image]
[image]

closed 2024-05-15 by andre1araujo