andre1araujo/YOLO-on-PYNQ-Z2 #4: Connection to Jupyter Notebook and Ethernet

rodrigoruz, 2024-05-16:
Hello Andre, thanks a ton for your work and great documentation on this project. I am trying to learn more about the PYNQ Z2 platform and I am using this project as a starter to learn more about Vivado and other tools. I was able to successfully replicate your results by downloading the SD card image you provide, but I am trying to access jupyter notebooks and connecting my PYNQ Z2 board to the internet to upload some of the processed images to google cloud. For some reason, the jupyter notebooks that are supported on the board's original supported card image, are not available on the card image that I have downloaded from your project. I was wondering if you know the reason behind is or if you have any suggestions as to how I could approach enabling this to be able to upload some of the generated images from YOLO to GCP. Thank you in advance, Rodrigo.

andre1araujo, 2024-05-16:
Hello Rodrigo!
I'm glad my project and documentation are being helpful for you!
Beccause of lack of time, I could not reach the part of setting the Jupyter Notebook on the PYNQ-Z2 with the modified SD card image. However,this is how you might access the Jupyter Notebook on the board:

1. Connect your PYNQ-Z2 board to your internet router and configure your computer's IP accordingly. I went trough this doubt before on this [issue](https://github.com/andre1araujo/YOLO-on-PYNQ-Z2/issues/1).
2. On your computer (now connected to the board on the internet), you should browse `http://pynq:9090/` or `http://192.168.2.99`, but on the last one you should replace "192.168.2.99" with the actual IP address of the board.
3. Having access to Jupyter you can now upload any files you want!

The files that appear when you use the original SD card image do not appear with the custom SD card image because those files are more examples presented by Xilinx for the user to experiment. This custom image aims to implement the DPU with the right configurations and so it did not prioretize the example notebooks. However, you can download some of the original files on this [link ](https://github.com/Xilinx/PYNQ/tree/master/boards/Pynq-Z2)from the Xilinx PYNQ repository.

[image]

I don't know how you can upload some of the images to Google cloud but I am pretty sure that would involve the installation of some specific package/ library that you would call on the notebook so you could move it to your cloud. Take a look at [this ](https://stackoverflow.com/questions/60470095/upload-files-to-google-cloud-storage-bucket-from-google-cloud-datalab-using-pyth)and see if it helps. You might want to search something involving "upload to google cloud using jupyter notebook".

Other thing you have to take in consideration is that originally on this tutorial, you use YOLOv3 with a c++ script. On jupyter you have to use python on a Jupyter Notebook (.ipynb). I don't know how you could convert the c++ script to a jupyter notebook so I think the best thing to do is try to find some implementation already in ipynb. I found [this ](https://community.element14.com/members-area/personalblogs/b/ralph-yamamoto-s-blog/posts/dpu-pynq-on-ultrazed-ev)one which should be compatible with my work (Scroll down the website until you find the tutorial).

[image]

It might need some modifications so feel free to contact me for help!

I hope this information is helpful! Good luck on your projects! :)

rodrigoruz, 2024-05-16:
Hello Andre, thank you in advance for your thorough response. I read many suggestions to connect directly to the router, the problem is I am trying to access the internet through a local network in my office space, so I do not have access to a router. Did you find any ideas on how to go around this if you don't have access to a router? As a separate conversation, any pointers on how setting the Jupyter Notebook on the PYNQ-Z2 with the modified SD card image might be approached? Thank you in advance

andre1araujo, 2024-05-16:
Hello again Rodrigo.
The other option I know to connect the board to the internet without using the router is by using a Wi-fi extender. I haven't tried this method but [here on this guide](https://blog.umer-farooq.com/a-pynq-z2-guide-for-absolute-dummies-part-i-fun-with-leds-and-switches-47dd76abf9a9) the author mentions that option and redirects to a page that might help get that set up. 

[image]

These extenders are cheap as fas as I know so it might be a viable option.

Another option I read about was using a wi-fi dongle. [This documentation](https://pynq-testing.readthedocs.io/en/latest/appendix/usb_wifi_device.html) mentions some dongles you can use that are tested. There are quite some forums about this but I need to read it better to understand how that should work.

To access jupyter notebook, you should do the setup normally as [described here](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/deployment-on-pynq-z2/board-setup#preparation). But, on the Network connection part, you just do the connection to the router or wi-fi dongle or wi-fi extender. Then after everything is setup (including the USB connection), you just browse on Google or other browser the board IP address. To do this, you use this command `ifconfig eth0`, take a note of the IP address that appears and paste it on the browser like this: `http://192.168.2.99`, where "192.168.2.9" is the IP adress that appears on the command line that you took a note.
**This will give you access to the jupyter notebook even on the custom SD card image.** The password should be `xilinx `or `root`.

I hope this helps you. Feel free to ask any more questions!

closed 2024-06-03 by rodrigoruz