andre1araujo/YOLO-on-PYNQ-Z2 #5: pynq-z2 always restarts automatically

WShawnMing, 2024-05-26:
Hello Andre，After compiling the project, when I run 'yolo_img dog.jpg', my pynq-z2 always restarts automatically, I tried different image files, none of them worked. Is this the cause of insufficient resources?
[image]

andre1araujo, 2024-05-26:
Hello!
First of all, from the screenshot you attached I can see already one problem. You need to run the command `yolo_img dog.jpg` on the "yolo_pynqz2" folder and **not** "yolo_pynqz2_data". Forget about the "yolo_pynqz2_data" as this one is for obtaining YOLO metrics. It is not useful after all.
What you **should do** is, after log in, **acess the folder with this command `cd yolo_pynqz2` and then `yolo_img dog.jpg` to test the detections on the example image**.
If the PYNQ-Z2 completely crashes as you say, **that is probably related to power**. I suggested somewhere on the GitBook tutorial that as this application is very demanding in terms of power, **you should connect the power using the adapter** and not the USB. Take a look at the following image to see what I mean:

[image]

If you want to know more specifications about the Power adapter, I wrote some informations on [this part of the tutorial](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/requirements#hardware). Scroll down a little bit to see it.

On a final note, I should warn you that the DPU might crash in normal circumstances. Not crash the PYNQ-Z2 but the program. It will display a error about "DPU Timeout". I could not find the reason for this but it might not affect your applications.

Thank you very much and I hope this helps! :)

WShawnMing, 2024-05-27:
Thanks for your patient answer, the cause of the restart was indeed the power supply, after changing to a DC adapter, the restart issue was resolved and the procedure performed correctly, thank you again and have a good day! :)

closed 2024-05-27 by WShawnMing

andre1araujo, 2024-05-27:
You are welcome! Enjoy the project! :)