andre1araujo/YOLO-on-PYNQ-Z2 #7: Using Mobilenet

MaRcOsss1, 2024-07-02:
Hey Andre, I had a doubt regarding Model Conversion. I am trying to use Mobile net trained model, which is already based on Tensorflow, Do I need to undergo model conversion for that? I dont think I should , under that case what could be my next step? Should I move to quantization?

So do you know where I can refer to scripts for quantization? or atleast I can get Idea how to quantize my mobilenet model?

andre1araujo, 2024-07-02:
Hi!
If your model is already in tensorflow, **you can move to the quantization process directly**. I assume you have MobileNet model in a ".pb" format, so, you should move that file to the quantization folder.
**Important note:** DNNDKv3.1 is the quantization and compilation tool of Xilinx and that version supports `**Tensoflow 1.12**`! The model has to respect that version otherwise it will not work. There might be a way to downgrade the model if it is in Tensorflow 2 as most of the implementations. This is maybe the biggest limitation of the deployment of neural networks on the PYNQ-Z2 with this tool.
If you will be using MobileNet for image classification and not object detection, you have most of the quantizatization process like I describe on the tutorial. Follow everything correctly but you will need to change the calibration script. I don't think that task is too hard so you can try to use chatGPT for that. On the other hand there might be some work done on that regard before. Just try to look up for implementations of MobileNet or other similar classification models but implemented on simillar hardware, like the Ultra96 board. Also, the DNNDK packages you download has some examples that might be useful!
I hope this helps. Fell free to ask any questions if you find further problems! :D

MaRcOsss1, 2024-07-02:
Hey Andre,
Okay I will check that process myself. Also, if I plan to implement object detection using mobilenet what do I have to do differently? What could possibly change?

andre1araujo, 2024-07-02:
If you are talking about MobileNet, it is a classification model. It just tells you what is on the image. There is SSD MobileNet which is a model for object detection which has a base on MobileNet. SSD will later require a script in c++ that might be considerably different from the one I have for YOLO. In any case, there are certainly pre-made example scripts from other people or from Xilinx. When I have time I can looks into that.
Good luck!

MaRcOsss1, 2024-07-05:
Hey Andre,
 I was setting up the DNNDK and was installing the necessary packages : `pip install numpy opencv-python==4.7.0.68 scikit-learn==0.24.2 scipy progressbar2 keras==2.2.4` and this happened :
[image]
[image]

andre1araujo, 2024-07-05:
Hi!
I have PTSD of that first error because I spent more than two weeks trying to deal with it haha. But whan that happens is due to incompatibilities of the packages. In this case, I went and checked the opencv-python package and aparently it deprecated:

[image]

What you need to do now is change the version to what they suggest there `4.7.0.71`.
`pip install numpy opencv-python==4.7.0.71 scikit-learn==0.24.2 scipy progressbar2 keras==2.2.4`
If it does not work, experiment with the version before that, which is `4.7.0.66`.
If that does not work too, you will have to try and downgrade or upgrade the rest of the packages.
Give it a try and then update me if it worked so I can change the tutorial.
Thank you!

The second problem is because you are not running the script `convert.py` on the same folder as the picture. You first have to access the directory of the python scrip like this:
`cd /home/aryan/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion`
And then you do the conversion:
`python convert.py yolov3.cfg yolov3.weights /home/aryan/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.h5`

I hope this helps. Good luck!

MaRcOsss1, 2024-07-08:
so apparently the version 4.6.0.66 is working for now , I tired  versions from 4.7 to 4.10 none of them worked but this one is wokring as you can see : 

> aryan@aryan-X510UAR:~/Downloads/xilinx_dnndk_v3.1/host_x86/decent-tf/ubuntu18.04$ pip install numpy opencv-python==4.6.0.66 scikit-learn==0.24.2 scipy progressbar2 keras==2.2.4
> Requirement already satisfied: numpy in /home/aryan/anaconda3/envs/decent/lib/python3.6/site-packages (1.19.5)
> Collecting opencv-python==4.6.0.66
>   Downloading opencv_python-4.6.0.66-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (60.9 MB)
>      |████████████████████████████████| 60.9 MB 54 kB/s 
> Collecting scikit-learn==0.24.2
>   Using cached scikit_learn-0.24.2-cp36-cp36m-manylinux2010_x86_64.whl (22.2 MB)
> Collecting scipy
>   Using cached scipy-1.5.4-cp36-cp36m-manylinux1_x86_64.whl (25.9 MB)
> Collecting progressbar2
>   Using cached progressbar2-3.55.0-py2.py3-none-any.whl (26 kB)
> Collecting keras==2.2.4
>   Using cached Keras-2.2.4-py2.py3-none-any.whl (312 kB)
> Collecting joblib>=0.11
>   Using cached joblib-1.1.1-py2.py3-none-any.whl (309 kB)
> Collecting threadpoolctl>=2.0.0
>   Using cached threadpoolctl-3.1.0-py3-none-any.whl (14 kB)
> Requirement already satisfied: six>=1.9.0 in /home/aryan/anaconda3/envs/decent/lib/python3.6/site-packages (from keras==2.2.4) (1.16.0)
> Requirement already satisfied: keras-applications>=1.0.6 in /home/aryan/anaconda3/envs/decent/lib/python3.6/site-packages (from keras==2.2.4) (1.0.8)
> Requirement already satisfied: h5py in /home/aryan/anaconda3/envs/decent/lib/python3.6/site-packages (from keras==2.2.4) (3.1.0)
> Collecting pyyaml
>   Using cached PyYAML-6.0.1-cp36-cp36m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (677 kB)
> Requirement already satisfied: keras-preprocessing>=1.0.5 in /home/aryan/anaconda3/envs/decent/lib/python3.6/site-packages (from keras==2.2.4) (1.1.2)
> Collecting python-utils>=2.3.0
>   Using cached python_utils-3.5.2-py2.py3-none-any.whl (24 kB)
> Requirement already satisfied: typing-extensions in /home/aryan/anaconda3/envs/decent/lib/python3.6/site-packages (from python-utils>=2.3.0->progressbar2) (4.1.1)
> Requirement already satisfied: cached-property in /home/aryan/anaconda3/envs/decent/lib/python3.6/site-packages (from h5py->keras==2.2.4) (1.5.2)
> Installing collected packages: threadpoolctl, scipy, pyyaml, python-utils, joblib, scikit-learn, progressbar2, opencv-python, keras
> Successfully installed joblib-1.1.1 keras-2.2.4 opencv-python-4.6.0.66 progressbar2-3.55.0 python-utils-3.5.2 pyyaml-6.0.1 scikit-learn-0.24.2 scipy-1.5.4 threadpoolctl-3.1.0

andre1araujo, 2024-07-08:
Alright, thank you!
I will update the tutorial with that information. I hope everything will work accordingly for now on!
Good luck!

MaRcOsss1, 2024-07-08:
Hey Andre, I need to unzip the COCO dataset, right? It's taking forever to extract, so I just wanted to confirm. If it's not necessary, I won't waste so much time on it lol.

Also you mentioned in the tutorial that :
[image]
by "Images" you mean the 17GB folder right?

And what do I have to do with the quantization script given in the tutorial ? cause you have mentioned "before executing the quntization" and what and where is this "calibration function" ?
[image]

MaRcOsss1, 2024-07-08:
Hey, It showed Quantization ended only after 3/100
[image]

A folder anmed Quantization result is made but is empty:
[image]

And a pycache named folder is created which included a .pyc file:
[image]
[image]

andre1araujo, 2024-07-08:
Hello Aryan.
Sorry for the late response. I think you should first do this command to see the input nodes, output nodes and shape: `decent_q inspect --input_frozen_graph=model_data/yolo.pb`. I mention this on the tutorial and it is important, especially because you are using a different model. Complete the quant.sh script with the informations displayed by the command.
The COCO dataset shpuld be uncompressed. It takes some time, yes haha. The images are the 17GB I think.
Try to see if the names of the inputs and outputs of the network with the command and complete the script with the information. Then let me know everyting!
I am sorry if I take some time to answer but for now I tend to be a little more busy. I will do my best! :)
Good luck!

MaRcOsss1, 2024-07-09:
So I had ran that and I had updated the output nodes accordingly in the quant.sh, input nodes , output nodes everything is fine and right([here](https://github.com/andre1araujo/YOLO-on-PYNQ-Z2/issues/7#issuecomment-2214629042)). The problem is the quantization isn't getting completed, it stops at 1,2 or 3 (out of 100). Is it possible that my Computer isn't capable of doing that? ([here](https://github.com/andre1araujo/YOLO-on-PYNQ-Z2/issues/7#issuecomment-2214986368)) 

It's okay take your time, I can understand you can be busy and I appreciate you for still helping out.

andre1araujo, 2024-07-09:
It might be because of computational power. I am not sure but still, try to reduce the "calib_iter" to a lower number like 50 or even 10. Reducing the batch size generaly helps. You are also using the cpu so it is expected to be more demanding.
I hope this helps!

MaRcOsss1, 2024-07-10:
I have access to a better system but it runs on Ubuntu v20.04 , is it okay?
or should I install v18.04 on that system as well?

> Message ID: ***@***.***>
>

andre1araujo, 2024-07-10:
Yes, reducing the size of the batch helps reduce computational load. Normally computers with more computational power can use bigger batches, specialy with a GPU.
Unfortunatly, for DNNDK you can only work with Ubuntu 14.04, 16.04 or 18.04. Not Ubuntu 20.04. It says on the documentation here (page 8):

[image]

Good luck!

MaRcOsss1, 2024-07-10:
F okay. About the Batch size I reduced it to 10 instead of 50 and calib_iter as 20 and It worked, does it mean it takes 10 photos and then another 10 and so on, till 20, am I understanding this right or am I making a mistake?

Also I got on the step on compilation just to know the process but I ran into error as always :
[image]

andre1araujo, 2024-07-10:
Hi!
You are understanding it right. What you said is correct.
About the issue on compilation, it is a library that you need to install but I forgot to include it on the tutorial. 
Install it like this: `sudo apt install libgoogle-glog-dev`
I will now update the tutorial for that. I am sorry for all the problems you are encountering.
Good job!

MaRcOsss1, 2024-07-10:
Hardly any problems I encountered are on you, all of them are mostly on me
and to be honest upon our discussions from past 2-3 weeks all of them has
been solved so don't take it on yourself. When I encountered this error
today , First thing that came to my mind was what the heck did I do wrong
again haha.  Thank you so much Andre, you're the best.

andre1araujo, 2024-07-10:
No problem haha. 
This error in particular, somehow I still remember it and I quickly checked it on my notes and there it was. 
Keep pushing and any problem just tell me because you are helping a lot here.

MaRcOsss1, 2024-07-11:
Hey, is this warning serious? Or I should ignore it?
[image]

andre1araujo, 2024-07-11:
The warning you got is not very important. After all you just get a image with the DPU kernel. I don't know exactly how it is useful but maybe in some applications it might be. In any case, I will add that command. Thank you!
The rest of the process is on Windows indeed. Copy the files from the compilation to the Windows system and do the rest there.
I hope there are not much problems now!
Good luck!

MaRcOsss1, 2024-07-11:
without that library you won't get a .jpg file. I checked after the warning , and mine wasn't converted to .jpg file. After downloading that library, .jpg was generated.

andre1araujo, 2024-07-11:
Okay, thank you! Already updated the tutorials!

closed 2024-07-29 by andre1araujo

yasinozatli, 2024-12-28:
@andre1araujo Hello Andre, I hope you are doing well :). I have a problem, I kindly need your help.
When I run "yasin@yasin-VirtualBox:~/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion$ python3 convert.py yolov3.cfg yolov3.weights /home/yasin/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.h5" this command, it gives the error below. Thank you very much Andre.

None
Traceback (most recent call last):
  File "convert.py", line 262, in <module>
    _main(parser.parse_args())
  File "convert.py", line 245, in _main
    model.save('{}'.format(output_path))
  File "/home/yasin/.local/lib/python3.6/site-packages/keras/engine/topology.py", line 2556, in save
    save_model(self, filepath, overwrite, include_optimizer)
  File "/home/yasin/.local/lib/python3.6/site-packages/keras/models.py", line 103, in save_model
    with h5py.File(filepath, mode='w') as f:
  File "/home/yasin/.local/lib/python3.6/site-packages/h5py/_hl/files.py", line 427, in __init__
    swmr=swmr)
  File "/home/yasin/.local/lib/python3.6/site-packages/h5py/_hl/files.py", line 196, in make_fid
    fid = h5f.create(name, h5f.ACC_TRUNC, fapl=fapl, fcpl=fcpl)
  File "h5py/_objects.pyx", line 54, in h5py._objects.with_phil.wrapper
  File "h5py/_objects.pyx", line 55, in h5py._objects.with_phil.wrapper
  File "h5py/h5f.pyx", line 116, in h5py.h5f.create
OSError: Unable to create file (unable to open file: name = '/home/yasin/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.h5', errno = 2, error message = 'No such file or directory', flags = 13, o_flags = 242)

yasinozatli, 2024-12-28:
Hello Andre, I hope you are doing well :). I kindly need your help please. I get this error?
command --- yasin@yasin-VirtualBox:~/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion$ python3 keras_to_tensorflow.py --input_model=/home/yasin/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.h5 --output_model=/home/yasin/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.pb

 
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:493: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:494: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:495: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:496: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:497: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:502: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Using TensorFlow backend.
Traceback (most recent call last):
  File "keras_to_tensorflow.py", line 181, in <module>
    app.run(main)
  File "/home/yasin/.local/lib/python3.6/site-packages/absl/app.py", line 308, in run
    _run_main(main, args)
  File "/home/yasin/.local/lib/python3.6/site-packages/absl/app.py", line 254, in _run_main
    sys.exit(main(argv))
  File "keras_to_tensorflow.py", line 127, in main
    model = load_model(FLAGS.input_model, FLAGS.input_model_json, FLAGS.input_model_yaml)
  File "keras_to_tensorflow.py", line 62, in load_model
    model = keras.models.load_model(input_model_path)
  File "/home/yasin/.local/lib/python3.6/site-packages/keras/models.py", line 239, in load_model
    model_config = json.loads(model_config.decode('utf-8'))
AttributeError: 'str' object has no attribute 'decode'

yasinozatli, 2024-12-28:
I changed python3 to python. this error rises:
yasin@yasin-VirtualBox:~/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion$ python keras_to_tensorflow.py --input_model=/home/yasin/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.h5 --output_model=/home/yasin/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.pb
Traceback (most recent call last):
  File "keras_to_tensorflow.py", line 11, in <module>
    import tensorflow as tf
ImportError: No module named tensorflow
Thank you Andre.

andre1araujo, 2024-12-28:
Hello!
I am sorry for the delay on the response.
I came across that error before and I documented it on the GitBook tutorial. Here is what you need to do:
```
pip install pillow
pip install 'h5py==2.10.0' --force-reinstall
sudo apt install libgoogle-glog-dev
sudo apt-get install graphviz
```
It was a problem with the compatibility of Pillow and h5py packages.
Try doing this and then tell me if it worked! Thank you! :D

**NOTE:** I noticed you are using VirtualBox for that part but it will not work when you will use DNNDKv3.1 because it uses Tensorflow. You will need to use a computer with Ubuntu 16.04 or 18.04 installed. Sorry for the inconvenience. But if you really need I can help!

yasinozatli, 2024-12-29:
Thank you @andre1araujo , it works now. I have another problem,

(decent) yasin@yasin-VirtualBox:~/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion$ decent_q inspect --input_frozen_graph=model_data/yolo.pb

/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:493: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:494: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:495: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:496: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:497: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/home/yasin/.local/lib/python3.6/site-packages/tensorflow/python/framework/dtypes.py:502: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
Traceback (most recent call last):
  File "/home/yasin/anaconda3/envs/decent/bin/decent_q", line 5, in <module>
    from tensorflow.contrib.decent_q.python.decent_q import run_main
ModuleNotFoundError: No module named 'tensorflow.contrib.decent_q'

Could you help please @andre1araujo

yasinozatli, 2024-12-29:
Again hello @andre1araujo , because of errors, I restarted the Optimization and Compilation chapter. I have this error now:

(decent) yasin@yasin-VirtualBox:~/Downloads/xilinx_dnndk_v3.1/host_x86/decent-tf/ubuntu18.04$ pip install ./tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl
Processing ./tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl
Requirement already satisfied: numpy>=1.13.3 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (1.19.5)
Requirement already satisfied: grpcio>=1.8.6 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (1.48.2)
Requirement already satisfied: astor>=0.6.0 in /home/yasin/anaconda3/envs/decent/lib/python3.6/site-packages (from tensorflow==1.12.0) (0.8.1)
Requirement already satisfied: six>=1.10.0 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (1.17.0)
Collecting tensorboard<1.13.0,>=1.12.0
  Using cached tensorboard-1.12.2-py3-none-any.whl (3.0 MB)
Requirement already satisfied: absl-py>=0.1.6 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (1.4.0)
Requirement already satisfied: termcolor>=1.1.0 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (1.1.0)
Requirement already satisfied: wheel>=0.26 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (0.37.1)
Requirement already satisfied: keras-preprocessing>=1.0.5 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (1.1.2)
Requirement already satisfied: keras-applications>=1.0.6 in /home/yasin/anaconda3/envs/decent/lib/python3.6/site-packages (from tensorflow==1.12.0) (1.0.8)
Requirement already satisfied: gast>=0.2.0 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (0.4.0)
Requirement already satisfied: protobuf>=3.6.1 in /home/yasin/.local/lib/python3.6/site-packages (from tensorflow==1.12.0) (3.19.6)
Requirement already satisfied: h5py in /home/yasin/anaconda3/envs/decent/lib/python3.6/site-packages (from keras-applications>=1.0.6->tensorflow==1.12.0) (2.10.0)
ERROR: Could not install packages due to an OSError: [Errno 2] No such file or directory: '/home/yasin/.local/lib/python3.6/site-packages/numpy-1.19.5.dist-info/METADATA'

yasinozatli, 2024-12-29:
Thank you @andre1araujo , I made a mistake and had to restart. This time I get this error, could you please help?

(decent) yasin@yasin-VirtualBox:~/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion$ python3 convert.py yolov3.cfg yolov3.weights /home/yasin/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion/model_data/yolo.h5Using TensorFlow backend.
Illegal instruction (core dumped)

andre1araujo, 2024-12-29:
Hi!
Sorry again for the delay. The problem you are having now is likely related to the Tensorflow you are using. Try to make sure is Tensorflow 1.12 installed. But I don't think you can do this in Virtual Box. I mentioned on the tutorial that this needs to be done on a Ubuntu operating system and that VirtualBox does not emulate all CPU instructions (some of which Tensorflow uses).
In summary, I think the problem is that you are using a virtual machine and because it does not emulate the AVX instructions of the CPU you cannot proceed. You need to do that command on a Operating System not virtualized.

Sorry for the inconvenience :(
If you really need, I still have the Operating System on my computer and I might be able to do that part of the work for you.

yasinozatli, 2024-12-29:
Hi!
Thanks @andre1araujo, I tried tf 1.15 and it worked! But i have this problem now :( 
(decent) yasin@yasin-VirtualBox:~/Desktop/YOLO-on-PYNQ-Z2/DNNDK/model_conversion$ decent_q inspect --input_frozen_graph=model_data/yolo.pb
decent_q: command not found
Could you help please?

andre1araujo, 2024-12-30:
Hi!
Because you are no longer using Tensorflow 1.12 from the DNNDK application, you cannot use the "decent" tool. You have to install the wheel file like this:
```
cd Downloads/xilinx_dnndk_v3.1/host_x86/decent-tf/ubuntu18.04
pip install ./tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl
```
If it works let me know!

yasinozatli, 2024-12-30:
Hi @andre1araujo, I used your commands above and get the "Illegal instruction (core dumped)" error. This project will be running on zedboard and my term project. I need it absolutely, could you please help? I am grateful to you!

andre1araujo, 2024-12-31:
Hi!
Yes, I could help. Your problem is simply that you are using a Virtual Machine. You cannot do with a Virtual Machine.
I can do the process for you when I have the time. I will need your model (.cfg and .weights) file and also the dataset you trained the model on. If you could also provide as much details as possible I would appreciate it. For example, what is the aim of your project? Will you use YOLOv3 or YOLOv3-Tiny?
Good luck!

yasinozatli, 2024-12-31:
Hi!I am a beginner at FPGA. I saw your project and wanted to give it a try and have an experience. Also, I will make a presentation of this project at a school course. I want to use your YOLO model, is it possible? Thanks Andre!

andre1araujo, 2024-12-31:
Yes, you can use the model and averything. This project is public!
I see you are encountering some problems, so I can try and help you. If you could have a computer with Ubuntu 18.04, most of the problems will be solved but if not I can do that part for you. I need your model and training dataset!

yasinozatli, 2024-12-31:
Hi @andre1araujo, https://drive.google.com/drive/folders/1nUFkqvQK3zJCe2LYI6lUcLSqzE3osVOl?usp=sharing
I get these .cfg and .weights from https://pjreddie.com/darknet/yolo/ yolov3-416 as you mentioned earlier in the tutorial. I need a pretrained model, not trying to train. I dont have dataset. This model is trained by coco dataset. Purpose is to run yolov3 on Zedboard. Thank you for your support!

andre1araujo, 2024-12-31:
Ah, that is a lot simpler then! I already made all of that so you will only have to flash my SD card image on yours, insert it on your ZedBoard and do some aditional steps.

- First, you follow [this part](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/requirements) to have the micro SD card image.
- Then you follow the [deployment chapter](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/deployment-on-pynq-z2/board-setup). Everything here is the same for the ZedBoard. You just have to do the procedure for the ZedBoard.

If you come across any problem, just tell me and I will help. For now, just forget about the part you were doing and just follow the steps I told you here!
Good luck! :D