andre1araujo/YOLO-on-PYNQ-Z2 #8: Deploy on Other Development Boards

luminita02, 2024-07-04:
Hello,
Thank you very much for the tutorial you provided. However, my development board is not PYNQ-Z2 but ZedBoard. I have installed the PYNQ image on the ZedBoard. Can the tutorial you provided also be applied to deploy on other development boards? Thank you.

luminita02, 2024-07-04:
Thank you so much for taking the time to share. I truly appreciate your generosity in sharing your knowledge.
Thank you!

andre1araujo, 2024-07-04:
You are welcome!
I will keep this issue open for at least a mounth in case you find any problems so you can report back.
Whish you the best.

luminita02, 2024-07-05:
Sorry to bother you again..
When I followed the gitbook to the "Implementing the DPU on an SD card image" section, at the project building stage, specifically when entering the command "petalinux-build," the information shown in the picture appeared. Could you please tell me what impact this error will have on the subsequent deployment?
[image]
[image]
Thank you!

luminita02, 2024-07-05:
Hey Andre!
Thank you for your help. I have progressed to the "Deployment on Pynq-Z2" step, but I encountered an issue where the gcc and g++ libraries were not installed during the compilation process. I am not familiar with Linux development. I tried to install the gcc library but was unsuccessful and couldn't ping other address. I also attempted to download the gcc package to compile on the development board, but it was not successful. Have you ever encountered this situation? Do I need to compile gcc on the host machine and then transfer it to the Pynq system?
So sorry to bother you again, Thank you.
[image]

andre1araujo, 2024-07-05:
I am not sure why the gcc lybrary is not installed on the SD card image. Maybe it wasn't selected on the process.
Look at this step of the Petalinux on the tutorial:

[image]

I might have forgotten to include some package there related to gcc. I suggest you go to Petalinux, start it using `source /opt/petalinux/settings.sh` and then go to the rootfs configuration and look for a gcc package. If you can find it, please tell me and take a screenshot so I can update the tutorial.

I suspect that is the solution. I might have forgotten to include some othe installation.
I hope this helps and I am sorry for the inconvenience.
Good luck!

luminita02, 2024-07-06:
Thank you for your response. I've tried rebuilding the project. I found the gcc-related configuration in the configure section and checked it. Unfortunately, the problem still persists after recompiling.
[image]
[image]

luminita02, 2024-07-06:
Does this have anything to do with the board being used? I am completely following the tutorial from GitBook, and there is no step that specifies whether it should be ZedBoard or PYNQ-Z2.

luminita02, 2024-07-06:
![image](https://github.com/andre1araujo/YOLO-on-PYNQ-Z2/assets/98135191/c6a1cb45-ad19-4438-affd-9e19aac21e1
[image]
And I did not see any executable files related to g++ or gcc in the compiled rootfs.tar.gz folder.
Thank you for your patience and kindness!

luminita02, 2024-07-06:
Hello, I have resolved the issue of not including the gcc library. I speculate that we need to add the Filesystem Packages —> misc —> [*] packagegroup-petalinux-self-hosted step.
However, a new problem has emerged..
[image]

andre1araujo, 2024-07-06:
Thank you so much and I am sorry for the inconvenience!
That error seem to be related to the compiler not recognizing the ".elf" folder. That folder has instructions to the DPU to build the desired Keural Network. You will need to do the [Model optimization and compilation](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation) part in oreder to generate the ".elf" file needed for the ZedBoard in specific. 
At least that is what I suspect you have to do if you did not already do it.

Special attention to the [compilation part](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation/compilation) beacuse here you have to create a "dcf" file for your board. I actually think this is the only part you have to change to make the model work on the Zedboard.

I am sorry for the late response and for any inconvenience but I am a little bit busy these weeks and I might take some more time to answer. In any case, be sure to comment because I will always try to help.
I hope this helps with your problem.
Good luck!

luminita02, 2024-07-07:
Hey Andre!
Thank you for your help. However, the COCO dataset is too large, and the quantization process takes a very long time. Therefore, could you please provide the 'quantize_results' folder including 'deployey_model.pb', 'quantize_eval_model.pb' after quantization? I would like to go through the process. Sorry for the inconvenience.

luminita02, 2024-07-07:
And in the next section I found another probelm..
[image]
Perhaps the DPU IP that I used in Vivado project were out-of-date?
I downloaded the DPU IP from the internet, as it seems that it was not provided in your documentation.
Sorry for the inconvenience and thank you for your patience!

luminita02, 2024-07-07:
And you said "This file already exists for a limited set of Xilinx boards like the ZCU102 and the ZedBoard"
It's Zedboard when I created the Vivado project, but where can I find the .dcf file?
Thank you so much.

luminita02, 2024-07-07:
Do you have the DPU IP package? If so, could you please provide a copy? Thank you.

andre1araujo, 2024-07-07:
Hi!
The quantization results can be found on this [google drive](https://drive.google.com/drive/u/2/folders/1Z9f7k2nSqBmhzujIGrHJilvVg2uyP-CT).
I have ssen that error before on Dlet. In any case, [this is the link](https://drive.google.com/file/d/1-Ajpk6HIDHIkwDHdUJaB92HN5g_S0Bam/view?usp=sharing) that includes the DPU IP package. It is mentioned on [this tutorial](http://www.aiotlab.org/teaching/fpga/DPU%20on%20PYNQ-Z2_Vivado.pdf) (page 20) I followed and presented on my documentation.

When you download DNNDK v3.1, you have somewhere on that folder the .dcf file for the Zedboard. Try to look for it there!

[image]

I hope this helps you! Any further questions just feel free to ask!

luminita02, 2024-07-08:
Thank you very much for your help. I successfully converted the .hwh file to a .dcf file and used the quantized model you provided for compilation. However, the compilation result did not include the .jpg file you mentioned; instead, a .gv file was generated.

Additionally, when testing on the ZedBoard, an error occurred. I wonder if you could help me solve this problem. I apologize for the trouble.
[image]
[image]

luminita02, 2024-07-08:
And I tried to use the .dcf file provided by DNNDK
However, when I tried to compile, it seems that LeakyReLU is not supported.
So, I suppose we have to try the original method.
[image]

luminita02, 2024-07-08:
It is possible that there are issues with my generated .dcf or .elf files. I have tried regenerating the .dcf file several times, but the problem persists.
[image]
[image]
[image]
[image]
I'll try to include as many screenshots of the process as possible.
The last two images are of the .elf file opened with PyCharm. I don't know why they appear garbled at the end. So sorry for the inconvenience.

andre1araujo, 2024-07-08:
Hello!
I think you should use the .dcf file you have for the ZedBoard and not the one I used. The ".gv" file instead of the ".jpg" is not important because it is just to visualize the graph.
I don't know why the error accurs because the ".elf" file should have included the information of the layers. But appart from that, the rest of the compilation worked well. The problem seems to be on the ".elf" file that for some reason does not include all the information. The last screenshot you send me confirms that it does not include conv2d_67_convolution. Try to remove that line of code from th c++ script to see what happens.
In any case, maybe you should try [this project](https://github.com/ECSAlab/dnndk-zedboard) first.It if aimed for the ZedBoard and is to deploy the MNIST dataset for digit classification. It includes the ".dcf" file there!
I hope this helps!

luminita02, 2024-07-08:
Thank you for your answer.
Yes, the .hwh file was generated from the Vivado project I created, and the .dcf file was also regenerated by me. Thank you for your response. I will try the suggestions you provided.

luminita02, 2024-07-08:
Hello Andre!
It is possible that the .pb model I used was provided by you, and in the Model optimization and compilation section, there is a description about DNNDK installation where it is necessary to distinguish the type of development board. Therefore, I suspect that the problem might be here. I should not use the .pb file you provided.
[image]

andre1araujo, 2024-07-08:
Did you activate the Zedboard on that installation? If you didn't try the process again with my quantization results to see if that works. If it doesn't you might have to do it from the start :(
Just give it a try and let me know!

luminita02, 2024-07-08:
I tried using the .dcf and .pb files you provided to generate the .elf file, and the result was the same as before, with the error "Invalid Node name conv2d_67_convolution specified for DPU kernel yolo." 
I am planning to train my own model to try again.

luminita02, 2024-07-08:
I compared the .dcf file you provided with the one generated using the ZedBoard, and found that the two files are identical. The .elf file generated using the .pb file you provided also contains only 59 nodes.

I tried modifying vector<string> outputs_node= {"conv2d_59_convolution", "conv2d_67_convolution", "conv2d_75_convolution"}; to vector<string> outputs_node= {"conv2d_59_convolution"}.

The output is as follows: it seems that regardless of which experiment board is used, the .elf file obtained after compilation only contains 59 nodes.
[image]
[image]

Sorry to bother you again.

luminita02, 2024-07-08:
[image]
And I tried using Netron to open the deploy_model.pb file you gave me and found that this model has a total of 59 nodes. I'm not sure how this is related to the errors occurring when running the program.

andre1araujo, 2024-07-08:
I also don't know what the problem might be. But did you get yolo working? Is it satisfactory?
You might need to follow the wholde DNNDK process to make everything right. Quantization takes a lot of time because it uses the COCO dataset. If you plan on using YOLO to detect any specific object I would recommed you obtain the Tensorflow 1.12 compatible model file for YOLO on that specific object and do the process.

luminita02, 2024-07-11:
Hello Andre!
Thank you very much for your guidance. I tried using a different dataset for training and was able to get results. However, for an image with multiple targets, an error occurs during the run. Could you please provide some suggestions on this? Nevertheless, reaching this point is already very good.
Thank you!
[image]

andre1araujo, 2024-07-11:
Hello!
I am really happy for your progress!
The error you are encountering also happens with my normal implementation. I tried to look for it when I was doing the project but ended up not having the time to investigate further. It certainly can be solved with a library but it needs further investigation. In any case, it does not affect the application and you are able to get the results but with a uglier console print.
I am sorry I could not solve that problem. I hope as it is the application fulfills your objective. And if you have any more questions be sure to post them here!
Also, when your implementation on the ZedBoard is finished notify me so we can close this issue for other people to see in case they have the same question!

luminita02, 2024-07-12:
Okay, thank you for your help
Meanwhile, I would like to discuss the possibility of using a different network. If I want to modify the network structure or use another open-source network, how should I modify these steps? Do I need to obtain a new .weights and .cfg file? Additionally, when using a dataset for quantization, what modifications are required for the quantization procedure?

andre1araujo, 2024-07-12:
Hello!
Most of the steps keep the same if you want to try another model. You will have to change the "Model optimization and compilation" part using a different tensorflow model. For Tensorflow you are limited to the version 1.12 so you have to be carefull with that. You can use a Caffe model but you will have to change some things following the DNNDK v3.1 user guide. 
But if you want to use other models, I recommend using Tensorflow because it has more options but you have to keep in mind that version limitation (Tensorflow 2.0 and up are more common).
You don't need the ".weights" and ".cfg" file as those are specific of YOLOv3 Darknet architecture. Tensorflow models have the ".cfg" extension and that is the one you should use if you want to use other models from Tensorflow.

You can use any dataset you want for quantization. It is just for calibration but I think it should be the same the model was trained on. The quantization remains the same as all it really does is reduce the weight precision to fixed 8 bits.

I am sorry for any confusion but in summary, you can use Tensorflow 1.12 (max) and Caffe models. For tensorflow you istall the tensorflow configuration on DNNDK and for Caffe you install DNNDK with the Caffe configuration.
The process remains the same but you use a ".pb" file for the different Tensorflow model.
At the end, the c++ script executed on the board will have to face some modifications but you can look at examples online as many of the models were done before on other boards like the Ultra96 and Zedboard.

I hope this helps!

closed 2024-07-23 by luminita02