andre1araujo/YOLO-on-PYNQ-Z2 #9: Transplant the apt tool , Xilinx DNNDK intall failed

yolovino1, 2024-07-12:
Hello,Thank you very much for the tutorial you provided.However my board may be different from yours.Here is the documentation for my board.
Explanation about the factory image without APT software.Embedded Linux is generally a self-made Linux system specifically designed for certain scenarios, such as the factory image used in this document
Used to test the hardware functionality of the development board. APT software is a software installation tool for Debian based Linux distributions, while Redhat based Linux
Use yum as the software installation tool for the distribution version. These two Linux distributions are mostly used for desktop computers, but of course, there are also embedded versions for ARM
Imported version. Customized embedded Linux systems, in order to be lightweight and without the need for online installation of software packages, generally do not
I will use those Linux distributions. If readers need to use apt or yum, they can use Google to learn how to port the corresponding Linux distribution to open
On the hair board.
--------------------------
Factory image hasn't the tool of apt ,I can't use the command "make".

[image]

[image]

root@ATK-ZYNQ:~# uname -a
Linux ATK-ZYNQ 5.4.0-xilinx-v2020.2 #14 SMP PREEMPT Wed Apr 12 14:11:43 CST 2023 armv7l GNU/Linux
root@ATK-ZYNQ:~# cd /home/root/yolo-on-PYNQ-Z2/DNNDK/compilation/compile
root@ATK-ZYNQ:~/yolo-on-PYNQ-Z2/DNNDK/compilation/compile# make
-sh: make: command not found
root@ATK-ZYNQ:~/yolo-on-PYNQ-Z2/DNNDK/compilation/compile# sudo apt-get update
-sh: sudo: command not found

Last but not least,I use the seaships7000 instead COCO dataset,
[image]

[image]
[image]
[image]

[image]
[image]

andre1araujo, 2024-07-12:
Hello!
I think there has been a little misunderstanding there. First, can you tell me what development board you are using? You said it is not the PYNQ-Z2. We have to be carefull because it may not be compatible with this work!
Second thing I want to address is what version of Ubuntu you are using on the computer? Is it Ubuntu 18.04?
It seems that you managed to compile the model correctly with the seachips dataset. The next step is to organize the necessary files like this:
[image]

Then, you will connect your development board to the computer and pass the folder with all the files to your board. But you will have to install DNNDK v31 on the board and activate it for everything to work. That process might be a little different considering a different board.
If you can provide me those details I will try to help you better :)
Good luck!

yolovino1, 2024-07-12:
https://www.aliexpress.com/i/1005005697295252.html,
zhengdianyuanzi navigator zynq board fpga xlinx 7020 pynq linux ,xc7z020 board ,fpga is xc7z020clg400-2,
[image]

version of Ubuntu is the same to you , ubuntu-18.04.2-desktop-amd64.iso

andre1araujo, 2024-07-12:
Alright, so in any case, the board has a Zynq-7000  family chip. Also you are using Ubuntu 18.04 and everything should be the same.
My recommendation would be for you to follow [this part](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/deployment-on-pynq-z2/board-setup) of the tutorial but some things will be a little different. Just make sure you try to do more or less the things done on the PYNQ-Z2 in terms of connecting power, USB, ethernet,... The rest will be the same for your board. Follow that part and then you will have no problems. You can do it in Ubuntu or Windows if you want.
One note about your board, it seems excessively overpriced for what it is. I don't know if it has anything more than the Zynq chip but it seems too much. For a simillar price you could get a Kria SOM KV260 board wich is a lot more powerful. But that is just what I think and I am not saying this in any offense! 
I hope I could hel you. Any question you have just tell me!

yolovino1, 2024-07-12:
Ok,thanks you.Actually I cost it about 150$,but  I care about the problem how to achieve it,
I din't konw how to solve the problems which zynq7020_dnndk installation failed!
root@ATK-ZYNQ:/media/sd-mmcblk0p1/zynq7020_dnndk_v3.1# ./install.sh
Begin to install Xilinx DNNDK ...
Install DPU Driver ...
modprobe: can't load module dpu (extra/dpu.ko): invalid module format
Installation failed!
---------------
 Command " sudo apt-get update" and "make" couldn't be used.Therefore,I decide port the corresponding Linux distribution to open,but I didn't know how to search information for this theme.
root@ATK-ZYNQ:~# uname -a
Linux ATK-ZYNQ 5.4.0-xilinx-v2020.2 #14 SMP PREEMPT Wed Apr 12 14:11:43 CST 2023 armv7l GNU/Linux
root@ATK-ZYNQ:~# cd /home/root/yolo-on-PYNQ-Z2/DNNDK/compilation/compile
root@ATK-ZYNQ:~/yolo-on-PYNQ-Z2/DNNDK/compilation/compile# make
-sh: make: command not found
root@ATK-ZYNQ:~/yolo-on-PYNQ-Z2/DNNDK/compilation/compile# sudo apt-get update
-sh: sudo: command not found

Now there are only a few things left to complete the task.

andre1araujo, 2024-07-12:
Hi!
You only do the "make" command far down the process. You need to power up your board with the SD card image I provided, activate DNNDK with:
```
cd zynq7020_dnndk_v3.1
./install.sh
```
Then you will have to go to the folder with the model and execute the "make" command to compile the script!
[image]

Do you understand the process?
Good luck! :)

yolovino1, 2024-07-15:
The path to zynq7020udnndk-v3.1 is /media/sd-mmcblk0p1/zynq7020_dnndk_v3.1 
[image]

[image]

andre1araujo, 2024-07-15:
Are you running that command "./install.sh" on the board? Or are you using Linux Operating System on your computer?
Because you install that DNNDK tools on the board. You have the SD card installed with the image I provided [here](https://drive.google.com/file/d/1ETyM51KSWX_h1DVq9ptHPIux89oTrNPy/view?usp=drive_link) and inserted on the board. Then you have to access the board using MobaXterm or other X-11 capable software.
I need to know the first question to help you further.
Sorry for the inconvenience.

yolovino1, 2024-07-15:
I am sure that the FinalShell for the SSH connection of the board I am using is not executed on Ubuntu system. I do not know how to construct the SD card image, so I use the QSPI official factory image. When I start in SD card mode, I use BOOT BIN and image.ub cannot boot with SD card image. Does SD mode need to match the image of this card in order to boot and display in serial? I would like to ask if BOOT needs to be set before burning pynqz2.dpu_16.rar BIN and image.ub were imported into the SD card first, but the corresponding image could not be found. I have previously burned ZYNQ_7020uPYNQ, which can be recognized and displayed in the serial. Can I put the file in based on this? I'm not quite sure
----------------------------------------
root@ATK-ZYNQ:/media/sd-mmcblk0p1/zynq7020_dnndk_v3.1# ./install.sh
Begin to install Xilinx DNNDK ...
Install DPU Driver ...
dpu: disagrees about version of symbol module_layout
dpu: disagrees about version of symbol module_layout
modprobe: can't load module dpu (extra/dpu.ko): invalid module format
Installation failed!
root@ATK-ZYNQ:/media/sd-mmcblk0p1/zynq7020_dnndk_v3.1# uname -a
Linux ATK-ZYNQ 5.4.0-xilinx-v2020.2 #14 SMP PREEMPT Wed Apr 12 14:11:43 CST 2023 armv7l GNU/Linux
[image]
root@ATK-ZYNQ:/media/sd-mmcblk0p1/zynq7020_dnndk_v3.1# mmc0: card 59b4 removed
mmc0: new high speed SDHC card at address 59b4
mmcblk0: mmc0:59b4 LX32G 29.5 GiB 
 mmcblk0: p1
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
blk_update_request: I/O error, dev mmcblk0, sector 4098 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
FAT-fs (mmcblk0p1):mmc0: card 59b4 removed
mmc0: new high speed SDHC card at address 59b4
mmcblk0: mmc0:59b4 LX32G 29.5 GiB 
 mmcblk0: p1
 
FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

[image]
[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]

andre1araujo, 2024-07-15:
Hi!
You have to use this custom SD card image for it to work because this one has DNNDK libraries pre-installed. You take `pynqz2.dpu_16.rar` file, and uncomress it. Then you use BalemaEtcher for example to flahs it on the SD card. Follow [this part ](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/requirements#if-you-just-want-to-execute-the-yolo)of the tutorial.
After the image is flashed on the micro SD card image, you should insert it on your Board and follow the rest of the tutorial [here](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/deployment-on-pynq-z2/board-setup). You will have to keep in mind that first part that will be a little different on your board. But, if you read the documentation and follow the steps it should work on your board!
I hope this answers the question!

yolovino1, 2024-07-19:
[image]

andre1araujo, 2024-07-19:
Did you do it like in [this part](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/dpu-implementation/implementing-the-dpu-on-a-sd-card-image#sd-card-image) of the tutorial?
It seems right but there are two extra files on the BOOT partition. I am pretty sure they should not be there.
Also, did you check if your board had the jumper in SD card mode?
I am sorry it is not working. It should not have any problems. If you cannot do it, try tho download [this SD card image](https://drive.google.com/file/d/1ETyM51KSWX_h1DVq9ptHPIux89oTrNPy/view) and flash it on your SD card.
I hope this can help!

andre1araujo, 2024-08-19:
I will be closing this issue for inactivity. If any problem still persists or you need any help, be sure to open a new issue!
Thank you for understanding :)

closed 2024-08-19 by andre1araujo