andre1araujo/YOLO-on-PYNQ-Z2 #11: Inquiries about using Peta Linux

best100jr, 2024-08-04:
Thank you for the great article! I want to connect to the Internet by connecting a Wi-Fi dongle to the installed Peta Linux, but the iwconfig command does not exist, and even the normal sudo apt-get update does not work. What should I do in PetaLinux??  If you have any good articles on how to use Peta Linux, please recommend them please.

andre1araujo, 2024-08-04:
Hello!
Sorry for the late response. To connect to the internet, I suspect that the current SD card image should work.
In any case, you should try to look on [this previous issue](https://github.com/andre1araujo/YOLO-on-PYNQ-Z2/issues/4) where I also discussed the possibilities to connect the board to the internet.
If any of that does not work, please tell me. At least I can warn the users for that problem.
I do not have any experience with Petalinux but surely there is something that can be done there to support internet connection if it does not work.
Good luck! :)

andre1araujo, 2024-09-03:
Hi!
I will be closing this issue for inactivity. If you want to develop any other question be sure to open a new issue.
Thank you for understanding! :)

closed 2024-09-03 by andre1araujo