andre1araujo/YOLO-on-PYNQ-Z2 #12: Errors after running petalinux build

Mantra31, 2024-08-07:
I tried running the petalinux build command and these errors occured:

[image]

can you help me with this?

andre1araujo, 2024-08-07:
Hello!
You cannot build the petalinux project without making the configurations. Is that what happened? You just accessed the directory and made build?
If following [this part](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/dpu-implementation/implementing-the-dpu-on-a-sd-card-image#petalinux-project) of the tutorial led to that error, we have to investigate further.
It seems just that the program could not fetch one library from a URL. It leads me to suspect that your internet connection was not working correctly at the time because the URL exists and is functional.
I would try that again but making sure that the internet connection on the Virtual Machine is working right.
I hope this helps! If you make any progress let me know! :)

Mantra31, 2024-08-07:
Yeah, we tried again and figured that issue was internet and we have
progressed to further steps.

Thank you.

andre1araujo, 2024-08-07:
You are welcome!
If you find further issues, be sure to open a new issue as now I will be closing this one.
Good luck on the project!

closed 2024-08-07 by andre1araujo