andre1araujo/YOLO-on-PYNQ-Z2 #13: Model quantification problem

DurhackDa, 2024-08-19:
Hello author, thank you very much for sharing this project. When I ran decent_q inspect --input_frozen_graph=model_data/yolo.pb during the quantization stage, I got exactly the same results as you, as shown, 
[image]
but when I followed your steps and carefully changed the position of the dataset, When running the script, I keep encountering the following problem. Have you ever encountered this problem?
[image]

andre1araujo, 2024-08-19:
Hello!
I have not encountered that specific problem as far as I remember. In any case, from the error message, I suspect you are willing to use the GPU of your machine to accelerate the process, right?
If that is the case, you probably modified the `quant.sh` script placing a '1' on the GPU setting. The problem now is that you are probably using the CPU version of DNNDK. If you want to use the GPU version, you have to do [this pat](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation#dnndk-setup) again but using the gpu wheel file.
What I mean is that here, we did this command: `pip install ./tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl`
This command uses the "cp" meaning CPU. If you navigate the folder there is a GPU version you can use. I will check the name and give it to you in a second.

If you are indeed trying to use the GPU , give me that information because if not I have to search for other causes of the error.
Good luck! :)

DurhackDa, 2024-08-19:
I did not try to use the GPU. Did I use other versions of tensorflow when converting and generating yolo.pb, and then used version 1.12 in the quantization stage, which caused the content of yolo.pb to be unable to be read, leading to the current problem?

DurhackDa, 2024-08-19:
It is the same as what you are using in the project, but when I convert the yolo model from Darknet to Keras and convert the Keras model to Tensorflow, I may be using other versions of tensorflow.

DurhackDa, 2024-08-19:
Hello, I have made some modifications, but now I have encountered a new problem. When I use the dataset to train the model, I get the following error message, which seems to be due to insufficient memory. I would like to ask, how much memory did you use when you were training? Are there any rigid requirements?
[image]

andre1araujo, 2024-08-19:
Hi again!
My computer has about 12GB of memory. I was able to do it without any warnings. If the problem is indeed related to memory, you can just reduce the batch size for the quantization. That can be done by opening the calibration script called "graph_input_fn.py" and change the `calib_batch_size` to a smaller number. The default is 50, so you can experiment, for example with 30 or 20 and see if the error persists. 

[image]

If at batch size '1', the same error occurs it means the problem is related to something else.
I hope this helps you!
Good luck! :)

DurhackDa, 2024-08-19:
I just tried it and it still doesn't work. Maybe my available memory is too low, only 3G size. I'll try to see if I can complete the training task on a different computer tomorrow, thank you very much for your reply!

DurhackDa, 2024-08-20:
Hi, does this quantisation time have much to do with the number of images? I'm using 120 images to test the results first, is it possible that I don't need to wait as long as 12 hours.

andre1araujo, 2024-08-20:
The number of images of the dataset directly relate to the time it takes for the quantization process. It needs images for calibration, so if the number is lower it will take less time but on the other hand it will make the model less accurate.
The quantization time is also related to the processing unit, your CPU. Of course if it is a more powerful CPU it will take less time. This process would be considerably faster with a GPU than a CPU.
I suggest you just leave it at night working on the quantization. Make sure it is on a well ventilated room to make sure nothing overheats on your computer. You might also raise your laptop a little bit to let more air get in the intake vents.

DurhackDa, 2024-08-20:
My process is still terminated, can I use 20,000 photos and change calib_iter 100 to 20 and give up some accuracy.

DurhackDa, 2024-08-20:
I'm really sorry, I still can't finish the training, do you have a GENERATED QUANTIZE FILE that has been trained that I can use and continue with the project later?

发送自 Outlook for iOS<https://aka.ms/o0ukef>

[EXTERNAL EMAIL]

andre1araujo, 2024-08-20:
Yes, you can use the 20000 photos for the quantization and reduce the calibration batch size to help speed up the process.
Yes, I have the quantization results for you if you want to skip that step. You can access the files on [this Google drive](https://drive.google.com/drive/folders/11gCSzDKbSrAvArS232TQK5Ga-qBcJR5m). 
I hope this helps and sorry for any inconvenience!

DurhackDa, 2024-08-20:
Thank you very much for sharing. I will continue with the next steps and see what caused the failure later.

[EXTERNAL EMAIL]

DurhackDa, 2024-08-20:
[image]
I would like to ask where the zynq7020 dnndk v3.1 mentioned in the picture is? Sorry I didn't find it.

DurhackDa, 2024-08-21:
I found two small mistakes： 
(1) Copy the contents of the folder to the SD card here. 
[image]
There will be no zynq7020_dnndk_v3.1 folder here, just open root directly 
[image]
(2) The IP address here should be ping 192.168.2.99
[image]

andre1araujo, 2024-08-21:
Hi!
I did not understand correctly the first mistake you are pointing out. My English is not perfect at all haha.
In any case, the zynq7020_dnndk_v3.1 folder is located [here](https://github.com/andre1araujo/YOLO-on-PYNQ-Z2/tree/main/DPU%20implementation/Petalinux).
The second part I noticed it was 192.168.2.99 instead of 192.168.2.9. I already corrected it but it was no big mistake since both addresses should work!
Thank you very much for pointing out mistakes!

DurhackDa, 2024-08-21:
Hello, the first problem is that if there is * in this command, zynq7020_dnndk_v3.1 will not be created, and the file will not be found in the second command. I have a more major problem now. After I used deploy-model.pb to generate the elf file, there was only one result for image recognition in the end. I used NETRO to check the .pb file and found that it only had one solution, cov2d_59, and no 67 and 75, is this pb the result of complete training?
[image]

andre1araujo, 2024-08-21:
Hello!
What do you mean with "only one result for image recognition"? It only detects one class of objects?
I might have provided the wrong file but I am sure that is not the case. The .pb file was the result of 9 hours of quantization using the COCO dataset images.

DurhackDa, 2024-08-21:
Yes, it only detects the bicycle in the images

发送自 Outlook for iOS<https://aka.ms/o0ukef>

[EXTERNAL EMAIL]

andre1araujo, 2024-08-21:
It should work better than that. If you test it on other images, it will detect more objects. It just seems to be lacking the precision it normally has. 
Try instead to use [these compilation files](https://drive.google.com/drive/u/2/folders/1vjVQ8fHe-IJwtoLmnupZuPJlAxdrQD3b) I generated and see if it works better.

DurhackDa, 2024-09-02:
Hi, I would like to ask, in this label position in the figure, is this coordinate obtained under the original size of the validation set image or the size normalized to 416*416 size?
[image]

andre1araujo, 2024-09-03:
Hello!
Sorry for the delayed response.
**The coordinates are relative to the validation image with the original size** and not the resized image. The original image is resized to 416x416 pixels using the _'letterbox_image'_ function, sent to the YOLO neural network for the detections and then passes the _'correct_region_boxes'_ function to resize it back to the original size. Only then the _'deal'_ function is called, which registers the coordinates and objects.
I hope this helps!

andre1araujo, 2024-09-24:
Hello!
I will be closing this issue for inactivity. If you desire to continue reporting issues make sure you open a new one!
I am sorry for the inconvenience and  thank you for understanding :)