andre1araujo/YOLO-on-PYNQ-Z2 #14: camera

mark2146, 2024-09-03:
Hi, thank you very much for your help with the project. I’ve successfully used it, but I have two questions I'd like to ask. First, when using the camera, I’m experiencing disconnection issues, as shown in the image, and the image feed is very slow. Is this an issue with my camera, or is there something I can adjust because it seems our FPS are different? Second, is it possible to replace the YOLO model in the project with one that I have trained myself? Thank you
[image]

andre1araujo, 2024-09-03:
Hello!
I am sorry for the delayed response.

The problem related with the board suddenly shutting off is due to power issues. I recommended on the tutorial the use of a [external power source](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/requirements#hardware)  (a normal 12V transformer) because the application is very demanding and the USB power is more inconsistent. That should solve the problem completely.

About the latency of the application, it is expected to be very slow for a real time cenario. The DPU is responsible for processing the YOLO neural network which is very complex and has a lot of computations but the resources of the board don't allow for something faster, at least for now. The latency does not have anything to do with your camera!
I should mention that the FPS value might be different because I did not had time to change FPS count on the c++ program. I had a little error and the timer start should be placed somewhere earlier on the code.
If you want a faster application, I recommend you try to use Yolov2 which is a low less complex and should result on a better real time performance.

You can absolutely use your own model for this! If you trained your YOLOv3 model, you just have to do [this part](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation/convert-model-to-tensorflow) to convert it to keras, then to Tensorflow and follow the rest of the process. If your model has less classes it can actually be a little faster than the current version!

I hope this helps you! If you have any more questions feel free to ask on this thread or on a new issue! :)

mark2146, 2024-09-16:
If I understand correctly, you have successfully run the DPU and DNNK on the PYNQ Z2. Now you are wondering if you can use your TensorFlow model directly instead of running the YOLO neural network model. If so, is there a tutorial available? I have searched online and found that most resources are about CNN models, which are a bit too complex for me.Thank you

mark2146, 2024-09-16:
My idea is to use Edge Impulse to train a model for use on the PYNQ-Z2. However, since the PYNQ-Z2 does not support Edge Impulse directly, I have been struggling to find a solution. I noticed that the models trained with Edge Impulse are in TensorFlow format. Is it possible to apply these models directly to the PYNQ-Z2?（Edge Impulse is a tool for training TinyML models.）

andre1araujo, 2024-09-16:
Okay, so I looked up about the Edge Impulse services and specifically looked to [this Tutorial](https://www.edgeimpulse.com/blog/deploying-edge-impulse-embedded-ml-on-the-xilinx-kria-som/) for the Kria-som board. It seems that the application makes the training process for custom datasets really easy. 
That is all really good but unfortunatly I some limitations that will make it impossible to do it on the PYNQ-Z2. The problem is that it works with Tensorflow but I suspect is a recent version. Xilinx DNNDK stopped supporting the Zynq7000 series chips (the one on the PYNQ-Z2) some years ago and so my implementation is limited to the older Tensorflow 1.12.  As Edge impulse outputs a Tensorflow model on a more recent version, it cannot be done with DNNDK. 
The applications to embed CNN models on the Zynq7000 familly chips is quite limited nowadays as the industry is shifting to some more powerfull FPGAs with more resources.

mark2146, 2024-09-16:
So according to this theory, if I use TensorFlow 1.12 to train a model myself, will it be possible to use it on the PYNQ-Z2? I am currently working on a project and can't find many hardware-related resources online. Thank you for your help!

andre1araujo, 2024-09-16:
Absolutely! 
If you train a model in Tensorflow 1.12 or lower, you should be able to follow the tutorial and use it on the PYNQ-Z2 (accelerated on the FPGA). If you are able to get the model I can help you with the proceeding steps.
Good luck!

mark2146, 2024-09-18:
Thank you for your help. I will first use TensorFlow version 1.12 to train a small model. I’ll ask for your guidance when the time comes. I really appreciate it!

mark2146, 2024-09-19:
I'm sorry to ask such basic questions, but how did you successfully install TensorFlow 1.12? I downgraded my Python to 3.6, but I still cannot execute it successfully. It always reports that it's not supported. I also couldn't get the .whl file installation to work, and I haven't found a correct solution online. Could you help me with this?

andre1araujo, 2024-09-19:
Hi!
I did not install Tensorflow 1.12 directly. We work with DNNDK tool and it uses Tensorflow 1.12 to perform some of the operations.
I should note that you have to use Ubuntu 16.04 or 18.04 Operating System to make this work.
To install everything properly I suggest you follow [this part of the tutorial](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation#dnndk-setup). You can also scroll back to previous parts to make sure everything is right. 
There I show how to install Anaconda and other dependencies to make sure everything works correctly. To be sure, try to start from [here](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation).
The `tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl` file is the one that installs the Tensorflow.

I hope you understand!
Good luck!

mark2146, 2024-09-19:
If I understand correctly, you are using YOLO neural networks and then converting them into a TensorFlow format that can run on the PYNQ-Z2. But what I want to do now is to train my own neural network, for example using VGG19 with transfer learning, and then deploy it to the PYNQ-Z2. In that case, which version of TensorFlow should I use to train my neural network?

mark2146, 2024-09-19:
Thank you! I’ll continue to give it a try.

andre1araujo, 2024-09-19:
Sure, If you end up having more simple issues that require a lot of communication, you can message me on Linkedin [here](www.linkedin.com/in/andré-araújo-77a6882b0). All this can be a bit overwhelming for someone with not that much experience so I am here to help!

mark2146, 2024-09-19:
I have now trained my model. Should I [proceed ](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation/quantization)to quantize it and then deploy it onto the PYNQ-Z2 board? Since my model is already saved as a .pb file, can I skip the earlier steps? Also, do these steps need to be performed on Ubuntu, or can they be done on other operating systems? Could you please provide me with the Linked link again? I'm getting a 404 error on my end.Sorry for the continuous disturbance.

andre1araujo, 2024-09-19:
Sorry for the late response. I don't knwo how to share the profile haha. Can you give me yours and then I add you?

andre1araujo, 2024-09-20:
Well, we can try discord. Try to find it using "andrearaujo01".

mark2146, 2024-09-20:
I sent a invite.My name is 798!

closed 2024-09-24 by andre1araujo