andre1araujo/YOLO-on-PYNQ-Z2 #15: getting error in quantization

Afreenappu, 2024-09-13:
Hello author, I followed the given steps as it is, but I am still encountering some issues
after running sudo ./install.sh PynqZ2 i got output like this
[image]

1)when i run the command decent_q inspect --input_frozen_graph=model_data/yolo.pb I getting errors like this

[image]

2) by running the quantization script I encountered an error as follows
[image]
 can you please tell me the solution to resolve this issues

andre1araujo, 2024-09-13:
Hello!
The first screenshot indicates the correct installation of DNNDK on  your host computer. The rest of the problems seem to be related to the installation of tensorflow. It might be due to some packages not installed correctly so I suggest you repeat [this part of the tutorial](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation#dnndk-setup) to make sure that every installation happens correctly with no errors. 
If the problem presists it might be related to your Hardware. Are you using a Virtual Machine with VirtualBox for example? If that is the case, it is the problem, since Tensorflow cannot be installed with the .whl file since the virtualization of the computer does not bring some CPU special instructions. 
This part of the work has to be done on the Ubuntu operating system. You either do a dual boot on the computer or use a different computer with Ubuntu. 
I am sorry for any inconveniences! If you can provide your objective I can try to help further :)
Good luck!

andre1araujo, 2024-10-25:
Hi!
I will be closing this issue due to inactivity. If you need further assistance make sure to open a new one or just contact me.
Thank you for understanding! :)

closed 2024-10-25 by andre1araujo