andre1araujo/YOLO-on-PYNQ-Z2 #16: Questions About Mechanism of This Work

luminita02, 2024-09-24:
Hey Andre!
I am now looking to study the mechanism of this work. We have used DPU to achieve hardware deployment of the neural network, and I would like to understand how the data flows during the inference stage. My understanding is that the image is read in by the PS side, the data flows from PS to PL, and inference is then performed on the PL side, while the network's weight parameters are loaded onto the PL side.
Additionally, I would like to ask you another question. In the Aiotlab tutorial, do you understand how the hardware layout is designed? I followed the tutorial to complete the hardware layout design but do not fully understand its meaning. 
I have many questions I would like to ask, such as why a 150M clock and a 300M clock are being used, and why the DPU's address range is from 0x4F00_0000 to 0x4FFF_FFFF.
Sorry to bother you, and thank you!
[image]

andre1araujo, 2024-09-24:
Hello!
I think it is great that you are coming to look more deeply on the whole mechanism on the block design! I honestly have no clear notion of what is exactly happening on that design and so I can recommend you read the [DPU product guide](https://docs.amd.com/v/u/3.0-English/pg338-dpu) for detailed information. There they justify those reset and clock signals, the DPU interrupt and much more. They also specify that the minimum required memory space is 16MB but we can select the desired memory space, just like it was done when we selected addresses from 0x4F00 to 0x4FFF_FFFF.
If you want to know more details about the DPU, the product guide should be enough but you can also read [this article](https://ieeexplore.ieee.org/abstract/document/8891993) from the guys at Xilinx's Deephi department.
Personally, I have never used Xilinx block design except for that project so I don't have specific notion on how everything works. Xilinx has detailed information on the DPU but it is a pattented design with their specific architecture and dataflow. Changing that design might be very challenging.
I am sorry I could not help you much but I hope these guidelines can help you understand the whole system!
Good luck! :)

andre1araujo, 2024-10-25:
Hi!
I will be closing this issue due to inactivity. If you need further assistance make sure to open a new one or just contact me.
Thank you for understanding! :)

closed 2024-10-25 by andre1araujo