andre1araujo/YOLO-on-PYNQ-Z2 #18: Changing the dataset

MaRcOsss1, 2024-11-09:
Hi Andre, I hope you’re doing well! I’d like to get your advice on something. If I decide to change my dataset, what adjustments would I need to make throughout the workflow? Are there specific things I should keep in mind, and would I need to resize the training images to a particular size? And what changes might be required in the final execution script?

andre1araujo, 2024-11-09:
Hello again Aryan!
If you would want to use a different dataset, I suppose you also want to use a different number of classes to detect. For example, you could use a dataset for face detection. If that is the case, you would need the following changes on the workflow:
- On the model conversion to Tensorflow, you need to make sure you are using the YOLOv3 model but trained for the particular dataset and detection you want. For face detection you need to have the Darknet model or already the Tensorflow model with the parameters trained on the face detection dataset.
- On the quantization part, you would need to repeat the process with the desired dataset, instead of the COCO dataset. The input image size for the YOLO accelerator remains the same (416x416) and the callibration script already resizes the images of your dataset for you. You would also need to confirm the input and output node names to then use them on the quantization script, but that is well described on the tutorial.
- On the deployment part, you would only need to change the number of classes and the name of them on the c++ script. You would anso need to name the input and output nodes on the script but all of that is really easy.

Note: I think I have somewhere this project done for face detection for other student if you need it I can provide to you!

About the Tensorflow Lite part, I don't have a lot of knowledge about it. Assuming it has some tools to reduce the model size using quantization or prunning, I don't think it would be much help. For instance, DNNDK already quantizes the model so that the weights and activations use 8 bits (I can provide you the scientific paper about the DPU if you want). It is mandatory for it to use 8 bits because that is how the accelerator is configured. 
DNNDK does what we call "quantization after training" which is less effective than "quantization aware training" which is probably what Tensorflow Lite is able to do. 
Prunning the model could potentially help but would technically reduce the efficiency of the accelerator. Anyhow, it might help reduce the latency!
I would need more details about Tensorflow Lite to help you more on that. But in summary, if it can prunne the model parameters, it might help reduce the latency on the pynq-z2. Quantization should not be done.

I hope this helps in any way. If you need any help or the material I talked about on this post, make sure to tell me!
Good luck!

MaRcOsss1, 2024-11-10:
Okay, understood, I am myself exploring how to deploy tensorflow lite and
looking for ways to deploy it on the board. I was just experimenting with
the process and reviewing the scripts to see where adjustments might be
needed. Also, if you could, please share the face detection project you
mentioned, and send over the scientific paper on the DPU—just attaching the
link or paper name would be perfect.

andre1araujo, 2024-11-10:
Explore Tensorflow Lite and see if you can start by just prunning the existing model to see if that makes any difference. If such option does not exist or the Tensorflow Lite flow is more abstract, let it do the quantization normally and continue the process to deploy on the pynq-z2 to see if there us any difference in performance.
The article about the DPU is [this one](https://dl.acm.org/doi/10.1145/2847263.2847265). They also relaesed a pattent for the DPU and this article is essencially the same information. 
About the face detection project, It is available [here](https://drive.google.com/file/d/1vkuZ81Lqq0QNTAsLgoH8qUJY2pP14rMv/view?usp=sharing). The compilation is already done, you can just modify the c++ file with the class name "face" (only one class) and the input and output nodes. If I manage to find the file I will send you. Sorry for that inconvenience,

MaRcOsss1, 2024-11-11:
Thank you very much! Could you do one last thing to help me? By any chance,
do you have the generated annotations file mentioned in your metrics
section 2: detections-on-the-pynq-z2 for Tiny YOLO and YOLO? I’d like to
compare it.

andre1araujo, 2024-11-11:
Yes yes, I can certainly help!
You can consult the annotations using [this google drive link](https://drive.google.com/drive/u/2/folders/1GI-VVlpxcfqYh19qbeoHdnOPpmZWMiqn).
I am sorry for the delayed response but it took a lot of time to upload all those files.
I hope it helps!

MaRcOsss1, 2024-11-15:
Awesome, thank you so much Andre!

MaRcOsss1, 2024-11-15:
Hi Andre,

I’ve trained a YOLOv3 model on a different dataset, which is relatively
small, consisting of just 1000 images with 2 classes. The annotations for
this dataset are currently in `.txt` files (in the YOLO format), but I
noticed that your quantization process requires annotations in a `.json`
format, specifically following the COCO standard.
Yolo format : <class_id> <x_center> <y_center> <width> <height>
Could you advise on the best way to convert my `.txt` annotations to the
required `.json` format? If there are any tools or scripts that can help
streamline this process, I'd really appreciate your guidance. Thank you!

> Message ID: ***@***.***>
>>
>

andre1araujo, 2024-11-15:
Hi, Aryan!
Sorry for the misunderstanding, but now that I reviewd the process I am 90% positive you don't acctually need the annotations for the quantization process. It just needs the images. 
In any case, if you could test the quantization process to see if it works without the annotations, it would be much appreciated!
Let me know if you got it. Good luck!

MaRcOsss1, 2024-11-15:
Hey Andre, yes it doesn't need annotations i.e. the json file for
quantization, only images are needed. I have completed my quantization and
compilation for a custom dataset, I'll send you the files once I completely
deploy it on the board. I have used a custom fighter plane dataset, and
have trained for only 2 classes for now, which now I have successfully
quantized and compiled.

andre1araujo, 2024-11-16:
Good to hear!
Thank you and good job! :D

MaRcOsss1, 2024-11-18:
I’m encountering the same error I faced earlier, but today’s issue is particularly puzzling. Initially, everything was working fine—I successfully ran Tiny YOLO in real-time and on images. However, when I tried doing the same thing again later, everything stopped working.

The error first appeared when I used my custom dataset. I thought the issue might be with the object files, but then I swapped my custom `.o` object file (e.g., image.o) with the YOLOv3 COCO dataset object file. The code ran, but since the object file was different, it predicted everything except the planes I trained for.

**Here’s what I’ve done for my custom dataset:**
- I modified the image.cpp file to update the classes and labels.
- I adjusted the input and output nodes accordingly.
Despite this, the board still predicted labels from the COCO dataset, which is odd given that the labels and classes in my `.cpp` file were entirely different. This makes me suspect that the model was predicting COCO classes because I used the COCO` .o` object file.

Since `.o` files are generated by the Makefile, I assume I need to keep the objects folder empty in MobaXterm. However, the Makefile isn’t generating my `.o` files either—it’s throwing an error saying something like image.o doesn't exist.

When I tried running Tiny YOLO again, I faced the same error, even though it worked previously.

**Here’s what I’ve tried so far:** Formatting the SD card multiple times and copying the files back onto it.
**My questions are as follows:**
- For the custom dataset, what could be going wrong?
- You previously mentioned that my SD card image might be incorrect. How can I ensure it’s correct?
- Why would Tiny YOLO suddenly stop working after working fine initially?

```
root@pynqz2_dpu1:~# cd yolo_pynqz2
root@pynqz2_dpu1:~/yolo_pynqz2# make
make: Warning: File 'Makefile' has modification time 1200 s in the future
g++ -c -O2 -Wall -Wpointer-arith -std=c++11 -ffast-math -mcpu=cortex-a9 -mfloat-abi=hard -mfpu=neon programs/yolo_image.cpp -o objects/yolo_image.o
make: g++: **Command not found**
make: *** [Makefile:40: objects/yolo_image.o] **Error 127**
```

andre1araujo, 2024-11-18:
Hi.
If it worked previously, then the SD card image is the right one, and you have the DNNDK libraries installed. 
The object files are generated for each given c++ script. If you used the ones generated for the YOLO COCO dataset, then that is the expected result. 
I suggest you also delete the executable file, as well as the object files, and try to compile again with the Makefile. If it does not work, try a couple of times. 
By the gcc error, it is just a little confused with the object file. I might also suspect the object file name is not the same as mentioned in the Makefile. Also, make sure it is in the objects folder!
Let me know if it works!

MaRcOsss1, 2024-11-18:
sorry but which ones the exectuable file??
[image]

andre1araujo, 2024-11-18:
It is not mentioned on that image but after you compile the program successfuly with the Makefile, you will see it created a file called "yolo" or "yolo_image" without any extension. It will be present on the same folder as the Makefile and the dog.jpg image.

MaRcOsss1, 2024-11-18:
Hey Andre, everything is working okay now, tiny yolo is running, there was probably some issues with the zynq7020_dnndk_v3.1 file and it has been resolved now, but when I try to run my custom dataset, it working all but the board is not detecting anything. Do you know why would this happen? 

It's showing this
[image]

andre1araujo, 2024-11-18:
We need to make sure it is not the accuracy of the model so, could you try to make detections on images used for training? Pick one of the training images and see if it detects. Try with various training images.
I suspect it can be low accuracy and so it migh need more training. Do you obtained any mean average precision value after the training process? 
If the problem is not accuracy, then it might be something about the program. For instance, the image size on the output might be wrongly configured. When I did the project, some image sizes were a little weird when dsiplayed.

MaRcOsss1, 2024-11-18:
Ok, I'll take a look. I'll have to check. But the images used for training
were pf size 640x640 , but in the quantization we did it for 420x420, does
it make a difference?

andre1araujo, 2024-11-19:
Hi.
No, it will not make a difference because the YOLOv3 neural network has 416x416 images as input. So, on the training process you probably had to resize the training images. There are other YOLO versions on the site with different input image sizes but 416 seems to be the most common.
Sorry for the late responses!

luminita02, 2024-11-22:
Hi Andre, I hope you’re doing well! You are such a kind person. 
I have tried using different datasets (in my case, time series of communication signals), and I also used TensorFlow to build a network, trained it, and got a .pb file. I have successfully gone through the quantization step for the model, and the next step is compilation. However, when I tried to compile, I encountered the error in the picture:
[image]
I removed the Softmax layer from the model, so the warning about Softmax no longer appears, but the issue with [The value is not supposed!..] still exists. I reduced the size of the model—from the original 400KB to 200KB—but the same issue still occurs. And according to your tutorial, the YOLOv3 model has much more parameters.
I am curious if you have ever encountered this issue, or if you have any relevant documentation about using DNNDK. Thank you very much!

andre1araujo, 2024-11-22:
Hi!

I never came across that particular error. Errors like that are more common when we start using other custom CNNs because DNNDK was still in early development at the time, and they probably did not conduct enough tests. I would suppose that YOLOv3 is a very deep neural network, so it should be a good test to compress. 

In any case, I found [this issue](https://adaptivesupport.amd.com/s/question/0D52E00006iHifhSAC/dnnc-compile-failed?language=zh_CN) on the AMD forum and the comment redirects to [this](https://adaptivesupport.amd.com/s/question/0D52E00006iHiiqSAC/dnnc-error-outtype-intype?language=zh_CN) other forum question. If those are indeed related, then the problem is flatten and dense/FC layers on your network. I saw that sometimes, the DNNDK compiler takes initiative to execute the demanding FC layer on the CPU, but it probably happens when those are the last layers of the network. So you might avoid those (important) layers or try to replace them or convert them with a convolutional layer. Also take a look at the size, as mentioned in the comment.

I am sorry to report what seem to be the limitations of the DPU, but it really might be a little hard to deploy custom models. I once tried to implement the VGG19 model, and it had a lot of difficulties, even though it is a very well known model.

I don't know about the current progress on the Vitis AI framework, but now it might support a lot more layers. The problem is that, many times, we still don't have comprehensive tutorials for platforms like this. You might want to explore that if you cannot change the layers of your network.

I hope you can proceed, and I am sorry I could not fully help.

Good luck!

luminita02, 2024-11-22:
Thank you so much! I'll look into what you have referred!

luminita02, 2024-11-23:
Hi Andre!
Thank you very much for your help. It seems that the DPU doesn't support the Flatten layer. I have modified the model and completed the compilation, resulting in a new .elf file. However, during testing, I encountered an issue where the .elf file could not be loaded, with the following error:
[DNNDK] Fail to load DPU kernel: dpu_signal_0.elf, check if:
    1- Specified DPU Kernel name "dpu_signal_0.elf" is right.
    2- DPU Kernel "dpu_signal_0.elf" is compiled and linked into "/home/root/yolo_pynqz2/yolo_image" as expected.
    3- DPU shared library "libdpumodeldpu_signal_0.elf.so" is located in the system library paths.
I checked the name of the .elf file and it seems correct. Have you encountered this issue before?
[image]

andre1araujo, 2024-11-23:
Hi!
I still don't know why that problem happens but I have seen it before while helping someone. I just took some time to look on the messages and this is what he did to solve it:
- First you go to the MobaXterm and access the folder you want: `cd yolo_pynqz2`
- Then you execute the following commands: `gcc -shared -o libdpumodeldpu_signal_0.so` and then `dpu_tiny_yolo.elf`. This will create a .so file with the name they refer to on the error.
- Then apparently you go to "user/bin" and add the recently created .so file.
- If the error presists, you delete the .so file from "user/bin" directory.

If you could do this process and then report to me how it went it would be much appreciated!
I hope this solves the problem!
Good luck :)

luminita02, 2024-11-24:
Thank you very much for your help. I followed your method and successfully solved the problem. Also, in the .cpp file, the file name loaded for the .elf should be changed to signal_0 (the full name of my .elf file is dpu_signal_0.elf).
Thank you again for your help. I will proceed to modify the .cpp file, and I would appreciate any suggestions you have for writing the .cpp file.

andre1araujo, 2024-11-24:
I am relieved it worked for you!
About the .cpp file, that is the complicated part haha. I think that first you should consult the examples of the DNNDK v3.1 package you downloaded on Ubuntu. They had some small implementation examples of object detection and image classification tasks. You should look at the example according to your model task. For example, if your model does image classification, you should consult the ResNet50 image classification example they have. Other examples should be available online or specifically on GitHub if you search for it. 
With the gathered examples, you can try using an LLM to try and speed up the process of developing your own .cpp file. 
I am sorry I don't have specific files here to help you, but this part should not be completely difficult because you just have to integrate a program with the DNNDK libraries.
I hope this helps to guide you a little to the desired solution. If you have any more problems, just tell me and I will try to help as best as I can!

andre1araujo, 2025-01-21:
Hi!

I will be closing this issue due to inactivity. If anybody has any more questions about the topic feel free to re-open. 
Thank you very much! :)

closed 2025-01-21 by andre1araujo