andre1araujo/YOLO-on-PYNQ-Z2 #20: Questions Regarding the Quantization Process

MelWang416, 2025-01-21:
Hi Andre,

I hope you're doing well!

While following your steps during the Quantization part, I encountered an issue that I couldn’t resolve. I’ve already successfully downloaded DNNDK and converted my YOLO model to both yolo.h5 and yolo.pb formats using TensorFlow. However, I keep running into an error with decent_q as shown in my screenshot. Also, I’m unable to download the decent_q module using 'pip install' and 'conda install'.

Could you please help me figure out what might be going wrong here? Any guidance would be greatly appreciated!

Thank you so much for your time!

Best regards,
Mel

[image]

andre1araujo, 2025-01-21:
Hello Mel!

I see that the problem you are encountering is related to the "decent" instalation. The error suggests that but you also said you had trouble installing. I am guessing you are refering to this command:
`pip install ./tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl`

You need to make sure you follow the steps presented [here ](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/model-optimization-and-compilation#dnndk-installation)very closely because there are a lot of dependencies. There are also libraries that need to be installed later and you also need to make sure there are no errors there.

What I think is the problem in your case is that you are using Petalinux and it is on a Virtual Machine (VM). That cannot happen. As I mentioned on the tutorial, you need to have the Ubuntu operating system (OS) installed on the computer to make it work, since the VM does not use all the instruction set of the CPU. Tensorflow requires the AVX instructions of the CPU but the VM does not provide that.
So, the solution it to use another computer with Ubuntu 18.04 intalled or to dual boot your computer with that new OS.

I am sorry for the troubles and best of luck!
If you need any help I'm here to help!

MelWang416, 2025-01-21:
Hi Andre,

Thank you for your reply!

Yes, I’ve been facing issues installing TensorFlow using the command `pip install `./tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl` It consistently returns the `error:Illegal instruction (core dumped)`. After searching, I found that the issue is indeed related to AVX instructions, as you mentioned. Before your explanation, I was puzzled because my CPU supports AVX instructions, but it seemed like the Ubuntu virtual machine couldn’t utilize them properly. Thank you for clarifying this—it makes a lot more sense now!

So, if I understand correctly, I can proceed with the **DPU Implementation part** using the virtual machine. However, for the **Model Optimization and Compilation** step, I’ll need to switch to another computer running Ubuntu 18.04 or set up a dual-boot environment with the required OS. Is that correct?

I’m a beginner in this area, and your guidance has been incredibly helpful. Thank you so much for your help!

MelWang416, 2025-01-21:
Hi Andre,

Thank you for your reply!

Yes, I’ve been facing issues installing TensorFlow using the command `pip install ./tensorflow-1.12.0-cp36-cp36m-linux_x86_64.whl`. It consistently returns the error:` illegal instruction (core dumped)`. After searching, I found that the issue is indeed related to AVX instructions, as you mentioned. Before your explanation, I was puzzled because my CPU supports AVX instructions, but it seemed like the Ubuntu virtual machine couldn’t utilize them properly. Thank you for clarifying this—it makes a lot more sense now!

So, if I understand correctly, I can proceed with the **DPU Implementation part** using the virtual machine. However, for the **Model Optimization and Compilation** step, I’ll need to switch to another computer running Ubuntu 18.04 or set up a dual-boot environment with the required OS. Is that correct?

I’m a beginner in this area, and your guidance has been incredibly helpful. Thank you so much for your help!

andre1araujo, 2025-01-21:
Hi again Mel!

Yes, that is correct. Unfortunatly I faced the exact same problem back when I was developing my project. I know dual boot can be a scary process, especially for someone who is not 100% sure on what is doing but there are a lot of tutorials, including videos on how to do it online. The process can actually be reverted back later! If you plan on doing it, make sure to archive some of the most important files just to be sure.

I am glad I have been able to help you! I actually feel like I am a beginer too but I've been through this process so if you have any more problems feel free to report them here.

Good luck!

andre1araujo, 2025-02-14:
Hi! I will be closing this issue due to incativity. If you would like to ask another question feel free to reopen the issue or create a new one!
Thank you for understanding :)

closed 2025-02-14 by andre1araujo