andre1araujo/YOLO-on-PYNQ-Z2 #21: Doubt regarding modifying this project

vinilsaji2003, 2025-03-14:
Hi,
I have a question about modifying this project. I've retrained a YOLO V3 model for PCB fault detection using a custom dataset with six classes. We have obtained the best weights and the corresponding cfg file.

If we optimize and compile our model as described in your GitBook, can we implement it on the DPU provided in the SD card flash image for the PYNQ Z2?

If not, what changes are needed to make it compatible?

Regards,
Bigin Paulose
Vinil Saji

andre1araujo, 2025-03-14:
Hello!
That seems like an interesting project! 
I think you just need to follow the process again as it is. The only modifications are these:
- Use your PCB dataset for the quantization instead of the COCO dataset.
-  On the yolo_image.cpp program for running the YOLO on the PYNQ-Z2, you need to change the `INPUT_NODE` and `outputs_node` according to the result given by this command: `decent_q inspect --input_frozen_graph=model_data/yolo.pb` on the DNNDK quantization step.
-  You also need to change the variable `classification` from 80 to 6 since your model is trained on 6 classes. 
-  Finally, change the `class_names` to the name of your 6 classes in order. The class 0 name should come first on the vector.

[image]

Apart from these small modifications, I don't remember any other aditional step needed. If you find any problems on the process just come back to this issue and I will try to help!
I have helped a individual use this model for face detection (one class) and it worked fine so I expect your project to work well.

Good luck for your project! If it works please let me know because that is a very interesting application and I might feature it on this github page!

vinilsaji2003, 2025-03-14:
[image] 
Hey Andre,

Just wanted to say a big thank you for your GitBook and your help! We managed to get our YOLO V3 model running on the PYNQ Z2 with the DPU, and it worked great. Your guide made everything so much easier!

Really appreciate all the work you’ve put into this. It means a lot. Looking forward to your in our next venture. Thanks again! 😊

Cheers,
Vinil Saji
Bigin Paulose

closed 2025-03-14 by andre1araujo