andre1araujo/YOLO-on-PYNQ-Z2 #22: Yolo V3 custom training

vinilsaji2003, 2025-03-26:
Hi Andre,
This is Vinil Saji,I hope you remember me,we did the pcb fault detection project.Now I have run into a problem and I think you can help me

Basically the tiny yolo v3 model training for pcb defect detection was done on Google Colab,since you have only a limited runtime on Colab we had to run it for days until we obtained a model,but it had an mAP of only about 40 percent,we ignored that and went on with the hardware implementation

Now as we presented some faults are not getting detected because of the low accuracy ofcourse

Do you have an absolute appropriate resource or the code to be run in colab(alternative code different from our's--https://youtu.be/2_9M9XH8EDc?si=iJsfzMv860uWafmX) to train the yolo v3 model because we can't find an appropriate resource,It would be helpful because we have final year presentation in a few days time and the only thing that is separating me from an output is the model's mAP

Regards,
Vinil Saji

andre1araujo, 2025-03-26:
Hello again!

Sorry but because I did not need to train a custom YOLOv3 model on my project, I don't have any resource code for the training process.
But I am not understanding well what you actually need. Do you need to obtain the mean average precision (mAP) score of your trained model? Do you want the mAP of the model on the PYNQ-Z2?

Please develop more your request so I can try to help you better :)

vinilsaji2003, 2025-03-26:
The model i trained has a very low mAP,due to this the model when deployed on hardware doesn't detect some defects

I want to train another yolo v3 model with a higher mAP so that it provides more accuracy when implemented on pynq z2 hardware

I have trained model only on Google Colab but it has limited gpu run time

I just want an alternative method to train the yolov3 model so that I can obtain the weights of the model with a higher mAP, I think this increase in precision would ensure that more defects are detected when implemented on hardware

Regards, 
Vinil Saji

andre1araujo, 2025-03-26:
Alright, now I understand very well what you want.

The problem is that to train the model you need at least one GPU. It inevitably takes a lot of time to train the model because it is somewhat complex. If you train the YOLOv3-Tiny it can be a lot faster because it is a smaller model. If you don't have a GPU on your computer, you cannot train the model in time for the presentation. The only other option is to use google collab GPU for free or to pay to a cloud server service.
If you will train the model, I can recommend you train it with 8 bit fixed point format instead of the 32 bit float point format. It migh help gain more detection accuracy.

Unfortunatly I don't have the resources nor the code for that. You can try to use a LLM like ChatGPT to make a pytorch, tensorflow or Darknet script to train the model in python. If you will try this I suggest using [Claude](https://claude.ai/new). It usually gives better results.

In summary: If you want to train the model to have better accuracy, I advise to train the YOLOv3-Tiny because its faster. If you don't have a GPU on your computers, the most viable option is still google Collab. I also don't have a GPU so I can't do it myself :(

I am sorry I could not help on this. I will try right now to get a python script for training the YOLOv3-Tiny. **Can you tell me or give me a link to the dataset you used for PCB defect?**

vinilsaji2003, 2025-03-26:
Thank you Andre for replying,
I have been training the model on google colab, but every time I did it I encountered some problem like cuda version not compatible etc, Also i dont have a paid version of colab, so my gpu runtime is only 3.5 hours, but the training takes more than that, so I had to stop it midway many times , somehow i was able to solve the issues and that's where I got the 40 % mAP model,I am not able to make progress beyond that, It would be nice of you to help
The link to the dataset I used is - https://www.kaggle.com/datasets/akhatova/pcb-defects

andre1araujo, 2025-03-27:
Hello again!

I am sorry to report but I could not successfully train the model. Several problems came on the way and unfortunatly I just don't have the time to solve them. I was doing the training but for once, I was using YOLOv3-Tiny from Ultralytics which apparently is a little different, then I applied quantization aware training but it seems to be the default option with mixed precision and not INT8. Finally, I would also need to export to Keras format at the end. In essence, this needs some attention I cannot give right now because I have work from my course to do yet. 

When I tried to implement other models using DNNDK I usually had trouble. YOLOv5 I think is considerably bigger model so it may not fit the DPU. Maybe if you trained YOLOv5n or YOLOv5s it would be possible to deploy. The process should be the same but, as you mentioned, some layers may not be supported by the DPU, specially this version. 

Sorry for the information. I don't know the full context but when you present the work you've done, normally it's ok when the results are not so good. My work also was not perfect but it worked more or less. You can make suggestions of future improvements.

I will be glad if I can help in any other way. Good luck!

vinilsaji2003, 2025-03-27:
Hi Andre,
I am glad you tried to help,I can understand your academic pursuits.

Could you please share the python script that you used,it would be helpful.

Regards,
Vinil Saji 
Bigin Paulose

andre1araujo, 2025-03-27:
Hello Vinil and Bigin.

Yes, I can share the document. It is in txt format. You just copy and past on a code cell on collab. It is in python.
First, you need to go to account settings on kaggle and get an API key. It will be downloaded a kaggle.json file. When you execute the code with the GPU activated on collab, there will be a button asking you to select a file. You press it and select the kaggle.json file. This is to access the kaggle dataset.

I hope this helps!

[collab_YOLOv3-Tiny_training.txt](https://github.com/user-attachments/files/19484657/collab_YOLOv3-Tiny_training.txt)

vinilsaji2003, 2025-04-02:
Hi Andre,

I gave it my best shot, but unfortunately, it was all in vain. Numerous challenges arose, and with time slipping away, resolving them proved impossible.

That said, I’m willing to give it one final try! By any means, do you have any connections who might be able to provide the .weights file for the specific PCB defect dataset? Your help would mean the world.

Regards,
Vinil

andre1araujo, 2025-04-02:
Hello again!

I am sorry but right now I don't have anyone that could do that for you, especially considering you have limited time. You can try and look for this [repository](https://github.com/qfzack/YOLOv3-pytorch-PCB/tree/master) where a YOLOv3 is trained on Pytorch for PCB defect but using a different dataset. The problem once again is having a available GPU. Or you can try and contact the individual to ask for the .weights file.
If I had a GPU I would be able to take a day off to try and get a model for you. Still, you got some results so you will not end the project empty-handed! Understande that in embedded systems like this is expected to have low accuracy. We need a lot of time to optimize and make tests to bring the accuracy up but it requires GPUs. It's unfair but it's how it works unfortunatly.

It hurts that I could not help you, so I jut hope you will be fine with what you have.

Good luck!

vinilsaji2003, 2025-04-02:
Dear Andre,

We truly appreciate all the support and effort you've given us. Please don’t feel sorry about not being able to assist in one particular scenario—your contributions have been invaluable, and we couldn’t have done it without you.

We hope to collaborate with you again on another exciting venture in the future!

Best regards,
Vinil Saji 
Bigin Paulose

vinilsaji2003, 2025-04-08:
Hey Andre,
Just a quick question,did you convert the coco dataset images to the yolo v3's standard 416*416*3 before quantization and calibration

andre1araujo, 2025-04-25:
Hello again Vinil and Bigin!

I hope you suceeded on your project! I will be closing this issue for inactivity but you can re-open it or start a new one if needed. Don't hesitate in asking if you need any help! :)

Thank you!

closed 2025-04-25 by andre1araujo