andre1araujo/YOLO-on-PYNQ-Z2 #23: Erro na etapa "Real-time object detector"

PokaChips, 2025-06-25:
Olá André,

Antes de mais nada, muito obrigado por criar um projeto tão bom e por elaborar um tutorial tão detalhado.
Gostaria de abrir este issue porque encontrei um problema no processo “Real-time object detector”.

Sou coreano, mas achei que seria mais confortável para você ler este issue em português, então usei o GPT para traduzi-lo.
Portanto, caso haja algum uso inadequado da linguagem, peço sua compreensão. Obrigado.

No processo “Real-time object detector”, conectei uma webcam USB ao PYNQ-Z2 e editei o Makefile conforme o tutorial, executando o comando make.
Tentei cerca de 10 vezes, mas continuo recebendo o seguinte erro:

[image]

[image]

Em particular, quando executo pelo MobaXterm, após o erro a conexão é encerrada forçosamente.
Será que esse problema está na minha webcam?

Obrigado.

andre1araujo, 2025-06-25:
Hello!

I really appreciate your effort but you can write the comment in english if it gives you less effort :)
I will respond in english so the translation is potentially better.

I don't know exactly what the problem might be on this case but what I immediatly suspect is that the power supply to the board is not enough for this application. Somewhere on the tutorial I mentioned that it is better to use a power supply that connects to the board in **12 Volts** and that using the USB connection alone sometimes is not enough to power the DPU and it is prone to crash more often.
**Are you using the USB connection to power the board**? If you are, can you **try using a power adapter** like described on [this section](https://andre-araujo.gitbook.io/yolo-on-pynq-z2/requirements#hardware)?

If this is not the problem contact me again and I will try and research a little more about it!
Good luck :)

PokaChips, 2025-06-26:
Thank you for your reply. I’ve been using both the USB connection and the 12V power adapter from the start. Could the webcam be the issue?

andre1araujo, 2025-06-26:
The error might be related to the camera. To debug this, firts run this command on the PYNQ-Z2 terminal `v4l2-ctl --list-devices`. Something like "/dev/video0" should appear. If something different appears, you have to update the `yolo_video.cpp` on line 414.
Let me know about this so we can further diagnose the problem!

closed 2025-07-28 by andre1araujo